phoenix-rtos/phoenix-rtos-project #3: Is there a build guide for this project example ?

rqbh, 2019-11-30:
When i run command './phoenix-rtos-build/build.sh all', psh compile failed, is this project tested in a new fresh system? i'm running ubuntu 16.04, and toolchain , stdlib installed success.

agkaminski, 2019-11-30:
Hi! Example project is under developement at the moment and will be ready really soon - perhaps on Monday already.

agkaminski, 2019-12-04:
Hello again. As you may have noticed, we finished preparing an example system image for qemu-i386 target. You'll find all instructions in the README file. Should you encounter any issue, please provide us with it's details and we will try to work out the solution.

rqbh, 2019-12-06:
got this problem

34603008 bytes (35 MB, 33 MiB) copied, 0.0186444 s, 1.9 GB/s
Checking that no-one is using this disk right now ... OK

Disk /home/kevin/phoenix-rtos-example/_boot/phoenix-ia32.ext2: 33 MiB, 34603008 bytes, 67584 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes

>>> Created a new DOS disklabel with disk identifier 0xacea2aa5.
Created a new partition 1 of type 'Linux' and of size 32 MiB.
/home/kevin/phoenix-rtos-example/_boot/phoenix-ia32.ext2p2: 
New situation:

Device                                                     Boot Start   End Sectors Size Id Type
/home/kevin/phoenix-rtos-example/_boot/phoenix-ia32.ext2p1 *     2048 67583   65536  32M 83 Linux

The partition table has been altered.
Syncing disks.
mkfs.ext2: invalid option -- 'd'
Usage: mkfs.ext2 [-c|-l filename] [-b block-size] [-C cluster-size]
	[-i bytes-per-inode] [-I inode-size] [-J journal-options]
	[-G flex-group-size] [-N number-of-inodes]
	[-m reserved-blocks-percentage] [-o creator-os]
	[-g blocks-per-group] [-L volume-label] [-M last-mounted-directory]
	[-O feature[,...]] [-r fs-revision] [-E extended-option[,...]]
	[-t fs-type] [-T usage-type ] [-U UUID] [-jnqvDFKSV] device [blocks-count]

Maxez, 2019-12-06:
It looks like you're using an old version of mkfs. Please upgrade util-linux package (containing mkfs) to version 2.31.1. You can download the package here: [util-linux](https://packages.ubuntu.com/bionic/util-linux)

pawelpisarczyk, 2020-08-24:
We are working on improving the project building. We add detailed instruction on Github and CI with binary snapshot generation will be launched soon for all supported targets. Sorry for the inconvenience.

closed 2020-08-24 by pawelpisarczyk