phoenix-rtos/phoenix-rtos-project #5: Error building toolchain in openSuse Leap 15.1.

kemonats, 2020-01-11:
When I ran the build-toolchain.sh script when creating the mpfr library, I received an error:

`configure: error: libgmp not found or uses a different ABI (including static vs shared).`

Cause:
libgmp.a is placed in the lib64 directory instead of lib.
Similarly with mpfr and mpc.
In other distributions, e.g. Debian, Archlinux, there is no error, libgmp.a libmpfr.a libmpc.a are placed in the lib directory.

To fix this, I improved the boot-toolchain.sh script.
Now the above libraries are created when gcc is compiled.

The script was also tested in Debian and Archlinux.
It will probably work in all distributions.

If there was interest I can present this script.

agkaminski, 2020-01-13:
Hi, yes, we're interested. It would be great, if you could submit a pull request with those changes.

agkaminski, 2020-01-14:
Thank you, your contribution has been merged.

closed 2020-01-14 by agkaminski