phoenix-rtos/phoenix-rtos-project #11: Linker lacks some support of parameters

milisarge, 2020-07-01:
LC_ALL=C ./phoenix-rtos-build/build.sh clean all
--- Preparing filesystem ---
--- Saving git-version ---
--- Building phoenix-rtos-kernel ---
CLEAN hal/riscv64
CLEAN lib
CLEAN vm
CLEAN proc
CLEAN test
CLEAN posix
CC  test/test.c                  
 ASM _init.S                   data=0      text=182  
CC  lib/rand.c                  
CC  lib/bsearch.c               
CC  vm/vm.c                    
 ASM _interrupts.S             data=0      text=528  
 ASM string.S                  data=0      text=498  
CC  lib/strtoul.c               
CC  proc/proc.c                  
CC  main.c                  
CC  programs.c              
CC  vm/zone.c                  
 CC  syspage.c                 
CC  lib/printf.c                
CC  lib/list.c                  
CC  vm/kmalloc.c               
 CC  hal.c                     
CC  lib/cbuffer.c               
 CC  spinlock.c                
CC  vm/object.c                
 CC  console.c                 
CC  lib/rb.c                    
 CC  interrupts.c              
 CC  timer.c                   
CC  test/rb.c                    
CC  test/proc.c                  
CC  test/vm.c                    
CC  syscalls.c              
CC  vm/amap.c                  
CC  posix/inet.c                  
CC  proc/file.c                  
CC  proc/cond.c                  
CC  proc/mutex.c                 
CC  proc/userintr.c              
 CC  pmap.c                    
CC  proc/name.c                  
CC  proc/resource.c              
CC  proc/ports.c                 
CC  posix/unix.c                  
CC  proc/msg.c                   
 CC  dtb.c                     
 CC  cpu.c                     
CC  vm/map.c                   
CC  vm/page.c                  
CC  proc/process.c               
CC  proc/threads.c               
CC  posix/posix.c                 
 CC  exceptions.c              
EMBED programs.cpio
Subsystem           | text    | rodata  | data
==================================================
hal/riscv64/code.a  | 19068   | 0       | 23584  
lib/code.a          | 4688    | 0       | 0      
vm/code.a           | 14490   | 0       | 1128   
proc/code.a         | 25242   | 0       | 1348   
test/code.a         | 3536    | 0       | 248    
posix/code.a        | 15216   | 0       | 148    
main.o              | 812     | 0       | 388    
syscalls.o          | 2372    | 0       | 4      
programs.o          | 728     | 0       | 0      
[... 766 lines cut ...]
/run/media/root/depolama/OS_calisma/phoenix-rtos-project/phoenix-rtos-ports/busybox/busybox-1.27.2//include/libbb.h:376:7: note: expected 'const char *' but argument is of type 'int'
  376 | char *xstrdup(const char *s) FAST_FUNC RETURNS_MALLOC;
      |       ^~~~~~~
/run/media/root/depolama/OS_calisma/phoenix-rtos-project/phoenix-rtos-ports/busybox/busybox-1.27.2/libbb/xfuncs_printf.c: In function 'xchroot':
/run/media/root/depolama/OS_calisma/phoenix-rtos-project/phoenix-rtos-ports/busybox/busybox-1.27.2/libbb/xfuncs_printf.c:409:6: warning: implicit declaration of function 'chroot'; did you mean 'xchroot'? [-Wimplicit-function-declaration]
  409 |  if (chroot(path))
      |      ^~~~~~
      |      xchroot
  AR      libbb/lib.a
  LINK    busybox_unstripped
Your linker does not support --sort-section,alignment
Your linker does not support --sort-common
Your linker does not support -Wl,--gc-sections
Trying libraries: m
Failed: -Wl,--start-group  -lm  -Wl,--end-group
Output of:
riscv64-phoenix-elf-gcc -Os -Wall -g -fomit-frame-pointer -fdata-sections -ffunction-sections -mcmodel=medany -fno-builtin -Wstrict-prototypes -I/run/media/root/depolama/OS_calisma/phoenix-rtos-project/_build/riscv64-spike/include/ -Dphoenix -Wall -Wshadow -Wwrite-strings -Wundef -Wstrict-prototypes -Wunused -Wunused-parameter -Wunused-function -Wunused-value -Wmissing-prototypes -Wmissing-declarations -Wno-format-security -Wdeclaration-after-statement -Wold-style-definition -fno-builtin-strlen -finline-limit=0 -fomit-frame-pointer -ffunction-sections -fdata-sections -fno-guess-branch-probability -funsigned-char -static-libgcc -falign-functions=1 -falign-jumps=1 -falign-labels=1 -falign-loops=1 -fno-unwind-tables -fno-asynchronous-unwind-tables -fno-builtin-printf -g -Os -fomit-frame-pointer -static -L/run/media/root/depolama/OS_calisma/phoenix-rtos-project/_build/riscv64-spike/lib/ -z max-page-size=0x1000 -o busybox_unstripped -Wl,--start-group applets/built-in.o archival/lib.a archival/libarchive/lib.a console-tools/lib.a coreutils/lib.a coreutils/libcoreutils/lib.a debianutils/lib.a e2fsprogs/lib.a editors/lib.a findutils/lib.a init/lib.a libbb/lib.a libpwdgrp/lib.a loginutils/lib.a mailutils/lib.a miscutils/lib.a modutils/lib.a networking/lib.a networking/libiproute/lib.a networking/udhcp/lib.a printutils/lib.a procps/lib.a runit/lib.a selinux/lib.a shell/lib.a sysklogd/lib.a util-linux/lib.a util-linux/volume_id/lib.a archival/built-in.o archival/libarchive/built-in.o console-tools/built-in.o coreutils/built-in.o coreutils/libcoreutils/built-in.o debianutils/built-in.o e2fsprogs/built-in.o editors/built-in.o findutils/built-in.o init/built-in.o libbb/built-in.o libpwdgrp/built-in.o loginutils/built-in.o mailutils/built-in.o miscutils/built-in.o modutils/built-in.o networking/built-in.o networking/libiproute/built-in.o networking/udhcp/built-in.o printutils/built-in.o procps/built-in.o runit/built-in.o selinux/built-in.o shell/built-in.o sysklogd/built-in.o util-linux/built-in.o util-linux/volume_id/built-in.o -Wl,--end-group -Wl,--start-group -lm -Wl,--end-group
==========
/opt/osdev/riscv-phoenix-toolchain/riscv64-phoenix-elf/lib/gcc/riscv64-phoenix-elf/9.3.0/../../../../riscv64-phoenix-elf/bin/ld: cannot find -lgloss
collect2: error: ld returned 1 exit status
Note: if build needs additional libraries, put them in CONFIG_EXTRA_LDLIBS.
Example: CONFIG_EXTRA_LDLIBS="pthread dl tirpc audit pam"
make: *** [/run/media/root/depolama/OS_calisma/phoenix-rtos-project/phoenix-rtos-ports/busybox/busybox-1.27.2//Makefile:717: busybox_unstripped] Error 1
```
```

pawelpisarczyk, 2020-07-28:
We are now working on userland for RISCV64 and we check this carefully.

nalajcie, 2020-10-22:
This was fixed in https://github.com/phoenix-rtos/phoenix-rtos-build/commit/b075a98bf0c8fc102b10d99aaef96e02ccd7472e

The riscv64 toolchain was configured incorrectly (it assumed that newlib libc would be used, thus requiring `libgloss` to be always available). To see the change You either need to recompile the toolchain or use docker-based build (newer toolchain is embedded in the docker image).

closed 2020-10-22 by nalajcie