phoenix-rtos/phoenix-rtos-project #22: Invalid HEAD of phoenix-rtos-ports

kemonats, 2020-10-04:
Missing commit:
[39ad44c](https://github.com/phoenix-rtos/phoenix-rtos-ports/commit/39ad44c34fccae97ddd7f0fefc87a82f154d442e)

agkaminski, 2020-10-05:
Thank you. Fixed in b2cc70e0c0b1e9251b926e4ce865f6741e6a9375

kemonats, 2020-10-09:
But HEAD is set to `c40abbd` and should be `39ad44c`

pawelpisarczyk, 2020-10-10:
Guys, sorry. Now it seems that it was corrected. Please check.

kemonats, 2020-10-11:
I tried to build for different targets and my conclusions:

1. ia32-generic
Ok
2. armv7a7-imx6ull
Ok ( after change `Darwin` to `Linux` in `mtd-utils`)
Why building `mtd-utils` only for `Darwin`.
3. armv7m7-imxrt106x
Stop at building `busybox`
4. riscv64-virt
Stop at building `busybox`

I fixed script headers in `_targets/` directory PR #24.

nalajcie, 2020-10-27:
Hopefully due to introducing CI building - future problems with missing commits/other build problems would be detected automatically.

If You would find any build problem not detected by the CI, please open a new issue.

closed 2020-10-27 by nalajcie