phoenix-rtos/phoenix-rtos-project #38: Error while creating a project for armv7m7-imxrt106x

kemonats, 2020-11-25:
Busybox compilation error.

commands:
```
git clone --recursive https://github.com/phoenix-rtos/phoenix-rtos-project.git
cd phoenix-rtos-project/
TARGET=armv7m7-imxrt106x ./docker-build.sh clean all
```
the message
```
/opt/phoenix/arm-phoenix/bin/../lib/gcc/arm-phoenix/9.3.0/../../../../arm-phoenix/bin/ld: cannot find crt0.o: No such file or directory
collect2: error: ld returned 1 exit status
Note: if build needs additional libraries, put them in CONFIG_EXTRA_LDLIBS.
Example: CONFIG_EXTRA_LDLIBS="pthread dl tirpc audit pam"
make: *** [/src/phoenix-rtos-ports/busybox/busybox-1.27.2//Makefile:717: busybox_unstripped] Error 1
```

nalajcie, 2021-04-22:
is this still a problem, I think c7774c68c545d7e8ad70916a30e129543a55d98e fixed that (by disabling building ports)?

nalajcie, 2021-04-28:
Fixed with c7774c6, if the issue still persists, please reopen.

closed 2021-04-28 by nalajcie