phoenix-rtos/phoenix-rtos-project #40: Problems with building plo for imxrt106x target on MacOS

MaciejPurski, 2020-12-04:
There is a problem on MacOS related to arguments to `stat` command:
[image]
MacOS uses BSD version of `stat`, which does not recognize the format parameter `-c`, which is present on GNU version. `stat -c` command is used in few places in `phoenix-rtos-project`:
[image]

agkaminski, 2020-12-04:
Perhaps we can change `stat -c "%s"` to to `du -s | cut -f1`?

MaciejPurski, 2020-12-04:
Unfortunately, on MacOS option `-b` is not recognised. Here is the man:
https://www.unix.com/man-page/osx/1/du/

agkaminski, 2020-12-04:
We could use `du $1 | cut -f1` if we can accept rounding to full 512 byte blocks. @HBuczynski do we need exact size there?

HBuczynski, 2020-12-04:
The exact files size is not needed. The file size defines the end of the program in memory and is later passes on to kernel in syspage. In my opinion, we can accept to round to `PAGE_SIZE`, which is 512 bytes.

agkaminski, 2020-12-04:
It is a coincidence that PAGE_SIZE is also 512 bytes ;)

gerard5, 2020-12-04:
Why not use "wc -c" as it is portable by all *nix, and will tell you the size in bytes.
$(wc -c < $1)

kemonats, 2020-12-04:
If the scripts are to be portable, in this case there are two solutions:

1. the simplest already proposed by gerard5
`wc -c < file`
2. using different options depending on the OS
Linux `stat -c %s`
Darwin `stat -f %z`
e.g.
```
OS=$(uname)
case "$OS" in
     Darwin) 
        STAT_OPT=-f %z
        ;;
     Linux)
        STAT_OPT=-c %s
        ;;
     *)
        exit 1
        ;;
esac
```
and use
```
stat $STAT_OPT file
```
The solution for `stat -c '%s` was proposed by me, but I didn't know that it should to work on `non-Linux` systems as well.

I have question for MaciejPurski:
How did you build using docker or native toolchain?

MaciejPurski, 2020-12-07:
> I have question for MaciejPurski:
> How did you build: using docker or native toolchain?
> .
> For me, building stops at busybox issue #38
@kemonats, I was building it using native toolchain for target armv7m7-imxrt106x.

PR #42 (2020-12-16): imxrt106x: Fix buildling on MacOS
Substitute non-portable `stat -c "%s" $1` with `wc -c < $1`.
Fixes #40.
closed 2020-12-16 by agkaminski