phoenix-rtos/phoenix-rtos-project #112: Copied symlink to file isn't visible as symlink (busybox shell)

damianloew, 2021-06-22:
[image]

nalajcie, 2021-06-22:
Symlinks support is currently ongoing and should be added in about week

damianloew, 2021-06-22:
Ok, so close the issue?

damianloew, 2021-06-22:
> Symlinks support is currently ongoing and should be added in about week
Ok, so close the issue?

nalajcie, 2021-06-22:
> Ok, so close the issue?

No, please keep it open. We will close it only after symlinks support will be mainlined.

damianloew, 2021-06-24:
I checked the mentioned pull request. The problem still exists (phoenix-rtos-filesystems commit dummyfs: fix lookup errno (EINVAL -> ENOTDIR))

nalajcie, 2021-06-24:
Thanks for the checks.

Please note that for symlinks to work at all You also need https://github.com/phoenix-rtos/libphoenix/pull/99

Upon more careful reading I see that the issue described **is invalid** - `cp` by default dereferences symlinks, If You need to copy a symlink - use `cp -d`.

Verified by hand that `cp` and `cp -d` work as expected with the changes in mentioned PRs (copes file contents, copies symlink).

Leaving issue open until both PRs would be merged.

nalajcie, 2021-06-24:
> I checked the mentioned pull request. The problem still exists (phoenix-rtos-filesystems commit dummyfs: fix lookup errno (EINVAL -> ENOTDIR))

Editorial note - If You paste GH link to commit - like `https://github.com/phoenix-rtos/phoenix-rtos-filesystems/pull/39/commits/0132ea319f2eaf163b323cd7ffa1689901a2f876`, You will get much nicer result: https://github.com/phoenix-rtos/phoenix-rtos-filesystems/pull/39/commits/0132ea319f2eaf163b323cd7ffa1689901a2f876

closed 2021-07-08 by damianloew