phoenix-rtos/phoenix-rtos-project #115: toolchain: add c++ support

nalajcie, 2021-06-23:
As of https://github.com/phoenix-rtos/phoenix-rtos-build/pull/69 we're building `g++` compiler, but `libstdc++` fails to build due to `libphoenix` headers (it's impossible to compile anything without additional options for specifying target architecture) - because of that the `configure` step fails:
[image]

nalajcie, 2021-07-21:
Due to recent changes in libphoenix full `libstdc++` compilation is nearer, now we miss some `ERRNO` values and basic ctypes, should be easier to fix:
[image]

badochov, 2023-08-02:
Isn't this issue completed by now?

nalajcie, 2023-08-03:
Indeed as of phoenix-rtos/phoenix-rtos-build#125 has been merged it is.