phoenix-rtos/phoenix-rtos-project #130: User stack corrupted on 'pc-ata'

jsarzy, 2021-07-06:
After the introduction of the stack canary target `ia32-generic` "non-deterministically" fails (one in a hundred or rarely).
[image]
To reproduce build project with 'test' and execute runner in a loop:
```bash
while ./phoenix-rtos-tests/runner.py -Tia32-generic -t ./phoenix-rtos-tests/initfini/ ; do :; done
```
Error also had occurred in Github CI but I re-ran job.

Maxez, 2021-07-16:
I've tried to reproduce the issue with no success so far. Every single initfini test succeeds (I left the test loop running for a few hours, I guess >10000 test runs finished).

My test setup:
1. Image build command (enabled DEBUG to run stack canary check): `TARGET=ia32-generic DEBUG=1 ./phoenix-rtos-build/build.sh core fs test project image`
2. The test loop: `while ./phoenix-rtos-tests/runner.py -Tia32-generic -t ./phoenix-rtos-tests/initfini/ ; do :; done`

Please, let me know if you found another way to make the stack canary check fail.

Maxez, 2021-07-21:
The issue is caused by scheduler interrupt coming in the middle of `threads_canaryInit()` (after setting `t->ustack` and before copying canary data). It could be reproduced every time by adding extra delay in this spot, e.g.:
```c
void threads_canaryInit(thread_t *t, void *ustack)
{
	if ((t->ustack = ustack) != NULL) {
                /* Add delay */
		for (volatile int i = 0; i < 0xffffff; i++);

		hal_memcpy(t->ustack, threads_common.stackCanary, sizeof(threads_common.stackCanary));
	}
}
```
Protecting stack canary initialization with spinlock solves the issue.

closed 2021-07-21 by agkaminski

nalajcie, 2021-07-22:
As this is now fixed - we can re-enable DEBUG builds in CI - @Maxez can You prepare PR with neccessary changes in  `build.project` for `ia32-generic`?

Maxez, 2021-07-22:
@nalajcie and I agreed that the DEBUG builds should be enabled by default for CI builds only. Therefore `DEBUG=1` should be defined in CI workflows instead of `build.project` files.