phoenix-rtos/phoenix-rtos-project #134: psh: tabulation autocompleting fails

niewim19, 2021-07-08:
When trying to autocomplete the path in `psh` (any path, existing or not longer than one character) the `psh` fails:

[image]

The error message appears due to `canonicalize_file_name()` returning NULL:
https://github.com/phoenix-rtos/phoenix-rtos-utils/blob/7c0743d429cde35a98fe67b35763805fff855fd9/core/psh/pshapp/pshapp.c#L580-L586

niewim19, 2021-07-08:
Wrong psh usage of `canonicalize_file_name()`

gerard5, 2021-07-08:
This is true for all of the targets that use `psh`, and due to https://github.com/phoenix-rtos/libphoenix/pull/99#issuecomment-867023226 , it seems like the `psh` part (which You are linking to) should be adapted to the recent changes.

nalajcie, 2021-07-08:
Some preliminary work was done by me which simply fixes the issue, but I think the whole code should be carefully read and tested to verify it's correctness:

```diff
diff --git a/core/psh/pshapp/pshapp.c b/core/psh/pshapp/pshapp.c
index aa1fd3a..b80047a 100644
--- a/core/psh/pshapp/pshapp.c
+++ b/core/psh/pshapp/pshapp.c
@@ -577,16 +577,23 @@ static int psh_readcmd(struct termios *orig, psh_hist_t *cmdhist, char **cmd)
                                        i = n - i;
                                        c = path[i];
                                        path[i] = '\0';
-                                       if ((fpath = canonicalize_file_name(path)) == NULL) {
-                                               fprintf(stderr, "\r\npsh: out of memory\r\n");
-                                               path[i] = c;
-                                               err = -ENOMEM;
+                                       /* allow_missing_leaf == 1 -> partial file names */
+                                       if ((fpath = resolve_path(path, NULL, 1, 1)) == NULL) {
+                                               if (errno == ENOMEM) {
+                                                       fprintf(stderr, "\r\npsh: out of memory\r\n");
+                                                       path[i] = c;
+                                                       err = -ENOMEM;
+                                               }
+
+                                               /* TODO: are other errors fatal? */
                                                break;
                                        }
                                        path[i] = c;
```

niewim19, 2021-07-09:
This solution:
 - does not produce error that exits `psh` (which is good)
 - does produce a minor bug where trying to autocomplete already complete path an additional `/` is added at the end. You get:
 ```
(0 tabulations): de
(1 tabulations): dev/
(2 tabulations): dev//
(3 tabulations): dev///
```

nalajcie, 2021-07-14:
Do we have a fix to this issue? It's a high priority one, as it breaks `BES` project currently.

If not, please provide stop-gap (disabling TAB-completion altogether) ASAP

nalajcie, 2021-08-20:
Please note that tab autocompletion doesn't work for symlinks to folders (they are treated as files instead of folders), steps to reproduce:
```bash
/bin/sh
mkdir /tmp/folder
touch /tmp/folder/file1
ln -s /tmp/folder /tmp/symlink
^D

ls /tmp/s[TAB]
```
[image]

There is also a related issue `ls /tmp/symlink` should show `file1` (`symlink` folder contents), but shows just `/tmp/symlink` instead:
[image]
seems like invalid usage of `stat`?