phoenix-rtos/phoenix-rtos-project #136: busybox: using diff with stdin causes exception

damianloew, 2021-07-08:
[image]

nalajcie, 2021-07-08:
Editorial comment:
- please provide steps to reproduce as a simple text (code block, to be able to ^C^V easily)
- please provide as simple reproduction path as possible (eg. `eval` was not needed there :p).

I solved the issue as a part of tracing other obscure libc errors, it would be great to add automatic test for this issue but I don't have time for that now.

Origin of the issue:
 - `diff` detects that it's not a real file but a buffer/stdin
 - `diff` creates temporary file (`mktemp()`) and copies all contents there (temp file is being open with flag `O_RDWR`)
 - than `fdopen` (with `mode = "r"` is being used to convert `fd` to `FILE*`)
 - as the file modes are not exactly the same, `fdopen` returns `NULL`
 - `diff` doesn't check the result of `fdopen`, just uses `fread` on it directly -> resulting in Page Fault

POSIX requires the fd passed to `fdopen()` to **allow** the desired mode, not to be exactly the same (so "r" for O_RDWR file is fine) - https://pubs.opengroup.org/onlinepubs/007904875/functions/fdopen.html

damianloew, 2021-07-09:
Right, with this PR that works correctly. I will follow the given advice in next issues. So what unit tests I should add?

nalajcie, 2021-07-09:
> Right, with this PR that works correctly. I will follow the given advice in next issues. So what unit tests I should add?

Regarding the tests - I was thinking of doing `open` with different flags and than checking if `fdopen` works on them. Also verifying correct errno codes should be done (like `EBADF`, `EINVAL`, etc.). It might be covered with POSIX test suite when we will start using it, so our internal tests are not a high priority right now and can be done in spare time.

closed 2021-07-09 by nalajcie