phoenix-rtos/phoenix-rtos-project #138: kernel: stack canary on MMU systems fails

nalajcie, 2021-07-09:
Recently introduced stack canary is not working well on MMU systems. It probably gives false positives on `ia32-generic` and completely break `imx6ull`:
 - there is a page fault in `hal_memcmp` used in stack canary checking when accessing address `0x7fff8000`
 - exception is not being printed as `process_dumpException()` uses `proc_current()` to obtain info about current thread/process
 - as `proc_current()` takes `&threads_common.spinlock` which is already taken in `threads_schedule()` the system hangs

As a stop-gap, `DEBUG` should be disabled in kernel on MMU platforms.

I would also consider invoking `proc_current()` as late as possible when printing exception to get at least partial information.

nalajcie, 2021-07-09:
#130 might have the same root issue.

agkaminski, 2021-07-09:
IMHO this issue applies only to imx6ull target. Issue #130 is different - it's not bad memory access, it's valid canary destruction detection

nalajcie, 2021-07-09:
> IMHO this issue applies only to imx6ull target. Issue #130 is different - it's not a bad memory access, it's a valid canary destruction detection

Ok, I thought on MMU platforms a page fault would happen if we would exceed the stack (assuming we enter unmapped address? I don't know the memory layout of the process) - so this either should be a false positive or a corner case (we're reaching top of the stack but not exceeding it?).