phoenix-rtos/phoenix-rtos-project #142: Rewrite old tests using Unity and harnesses

jsarzy, 2021-07-09:
In [pheonix-rtos-tests](https://github.com/phoenix-rtos/phoenix-rtos-tests) there are old tests that could be rewritten using Unity framework or python harnesses:
- [disk/test_disk.c](https://github.com/phoenix-rtos/phoenix-rtos-tests/blob/master/disk/test_disk.c) (benchmark)
- [fs/test_fcntl.c](https://github.com/phoenix-rtos/phoenix-rtos-tests/blob/master/fs/test_fcntl.c) (unit tests)
- [fs/test_fs.c](https://github.com/phoenix-rtos/phoenix-rtos-tests/blob/master/fs/test_fs.c) (benchmark/unit)
- [mem/test_malloc.c](https://github.com/phoenix-rtos/phoenix-rtos-tests/blob/master/mem/test_malloc.c) (benchmark/unit)
- [mem/pmt.c](https://github.com/phoenix-rtos/phoenix-rtos-tests/blob/master/mem/pmt.c) (unit - include into malloc tests)
- [mem/test_memmove.c](https://github.com/phoenix-rtos/phoenix-rtos-tests/blob/master/mem/test_memmove.c) (unit)
- [mem/tst_mmap.c](https://github.com/phoenix-rtos/phoenix-rtos-tests/blob/master/mem/test_mmap.c) (benchmark/unit)
- [proc/*.c](https://github.com/phoenix-rtos/phoenix-rtos-tests/blob/master/proc) (part of them can be rewritten as unit tests, some can be parsed by harness (for example thread tests with printing))
- [stdio/test_scanf.c](https://github.com/phoenix-rtos/phoenix-rtos-tests/blob/master/stdio/test_scanf.c) (unit)
- [stdio/test_str2num.c](https://github.com/phoenix-rtos/phoenix-rtos-tests/blob/master/stdio/test_str2num.c) (unit)
- [time/test_*time.c](https://github.com/phoenix-rtos/phoenix-rtos-tests/blob/master/time) (unit)

If you want to rewrite test just create PR and link to this issue. I'll try to track what have been done. Benchmark tests could be done using test harness with some fixed ranges of benchmark. I encourage to expand some tests (e.g. malloc).

nalajcie, 2021-07-23:
@damianloew @niewim19 would be nice if You could do it in some spare time :)

niewim19, 2021-07-23:
Roger that!