phoenix-rtos/phoenix-rtos-project #145: busybox: crash after pipe yes command output to head -c 9999 (or other large value)

damianloew, 2021-07-14:
[image]
To reproduce:
1. edit /etc/rc.psh in rootskel like that (for posixsrv run in background):
[image]
2. run system image
3. run /bin/psh /etc/rc.psh (for posixsrv run in background)
4. run /bin/sh
5. run yes abc | head -c 9999

damianloew, 2021-07-14:
That's occurs only when building for uart (not always)

closed 2021-08-06 by Maxez

Maxez, 2021-08-06:
Short explanation on what was the issue:
`yes` gets blocked on write to pipe (pipe buffer is full) while `head` has already read all the bytes it wanted. `head` terminates and pipe_close() gets called but it doesn't wake up `yes` which stays blocked on write to pipe.

The issue didn't show up when using pc-tty because writing to pc-tty is apparently faster than writing to uart16550 and pipe buffer never got full.