phoenix-rtos/phoenix-rtos-project #157: building toolchain fails on systems without /lib/cpp

xvuko, 2021-07-21:
In stage 2 during target-libgcc compilation compiler checks fail due to headers import errors. This results in checks being run on /lib/cpp and not stage1 compiler build earlier. 

This is example of stage1 check failing from `<BUILD>/gcc-9.3.0/build/arm-phoenix/libgcc/config.log`:
```
configure:4039:  <BUILD>/gcc-9.3.0/build/./gcc/xgcc -B$<BUILD>/gcc-9.3.0/build/./gcc/ -B<BUILD>/arm-phoenix/arm-phoenix/bin/ -B<BUIILD>/arm-phoenix/arm-phoenix/lib/ -isystem <BUILD>/arm-phoenix/arm-phoenix/include -isystem <BUILD>/arm-phoenix/arm-phoenix/sys-include    -E -traditional-cpp  conftest.c
conftest.c:13:2: fatal error: assert.h: No such file or directory
   13 | #endif
      |  ^~~~~     
compilation terminated.
```
Adding symbolic link <BUILD>/arm-phoenix/arm-phoenix/include to <BUILD>/arm-phoenix/arm-phoenix/usr/include fixes `No such file or directory` but results in `#error "unsupported architecture"`. I see in build_toolchain.sh  comment that this is known issue.

I am building toolchain using simple hack but this fallback to /lib/cpp seems like something that should be happening.

nalajcie, 2021-07-21:
@pawelpisarczyk this is the same issue You have on macOS, I'll try to look into that today/towmorrow. It might be that I've changed `#error` to `#warning` in libphoenix `arch.h` in local testing and than forgot it was needed.

nalajcie, 2021-07-21:
Edit: I see from my original compile logs that it also falls back to `/lib/cpp`, but on my OS it points to gcc-9 so the compilation succeeds eventually.

closed 2021-07-21 by nalajcie