phoenix-rtos/phoenix-rtos-project #166: busybox: 'ash: write error: EPIPE' when piping some large text variable into head command

damianloew, 2021-08-12:
[image]
To reproduce:

1. edit /etc/rc.psh in rootskel like that (for posixsrv run in background):
Screenshot from 2021-07-14 14-50-28
2. run system image
3. run /bin/psh /etc/rc.psh (for posixsrv run in background)
4. run /bin/sh
5. run text='yes abc | head -c 9999'
6. run echo "$text" | head -c 5

Maxez, 2021-08-17:
Code in `echo.c` which prints the error (`full_write()` sets errno to EPIPE):
```c
errno = 0;
/*r =*/ full_write(STDOUT_FILENO, buffer, out - buffer);
free(buffer);
if (/*WRONG:r < 0*/ errno) {
    bb_perror_msg(bb_msg_write_error);
    return 1;
}
```
Busybox applets (`echo` in this case) assume correct SIGPIPE signal behaviour which should terminate the process.
As a quick check I modified `proc_write()` in kernel so that SIGPIPE signal gets sent:
```c
err = proc_send(oid.port, msg);

if (err >= 0) {
    if ((err = msg->o.io.err) == -EPIPE)
	threads_sigpost(proc_current()->process, proc_current(), 13 /*SIGPIPE*/);
}
```
As a result default SIGPOST signal handler terminates the process and error messages no longer appear on the screen:
[image]

Please note that 'the quick check' probably isn't a correct way of sending SIGPIPE signal - I'll check how it should be done.

nalajcie, 2021-08-17:
Verified on host (linux) that the `echo $text` process is being killed by `SIGPIPE` (incomplete write) if the write size exceeds the kernel pipe buffer (65535 bytes by default), code to reproduce:
```bash
strace -f busybox ash -c 'text=$(yes abc | head -c 655360); busybox echo $text | busybox head -c 5'
```
[image]

I don't know if the `EPIPE` errno is being set in case of partial write (to be checked).

julianuziemblo, 2026-10-22:
we now have SIGPIPE in kernel on write, and so the output is as should be:
```bash
(psh)% busybox ash

BusyBox v1.27.2 (2026-10-22 13:31:46 UTC) built-in shell (ash)
Enter 'help' for a list of built-in commands.

~ # text=`yes abc | head -c 9999`
~ # echo "$text" | head -c 5
abc
a~ # 
```

closed 2026-10-22 by julianuziemblo