phoenix-rtos/phoenix-rtos-project #176: kernel: `setjmp` / `longjmp` broken on `rt1064` (at least on phoenix-rtos-kernel@4167188)

nalajcie, 2021-08-23:
Changes from PR https://github.com/phoenix-rtos/phoenix-rtos-kernel/pull/227 broke `setjmp` / `longjmp` on `rt1064` target.

To be investigated, for now **do not update `phoenix-rtos-kernel` submodule.

gerard5, 2021-09-02:
I will add to the investigation that there is the same problem on `imxrt1176`, locally I reverted a single commit https://github.com/phoenix-rtos/phoenix-rtos-kernel/commit/92f3645ef7687347f04ece500c9c0aa15ec2add7 and the ` test-setjmp` started working on `imxrt1176`, I also verified the `test-setjmp` on ` imxrt1064` and it works too.

jsarzy, 2021-09-02:
Yeap I also looked into it. When segment has LOAD type and RW flags we don't map it into memory, just uses ELF binary segment directly. So we don't relocate 2 entries from setjmp.

nalajcie, 2021-09-02:
@gerard5 thanks for narrowing it down to single commit, @jsarzy I didn't understand fully what it's about so decided to check it myself :).

My remarks:
1. The change done by @agkaminski is correct, but IMHO should be a fatal error (non-GOT relocations in `.text` are fatal sooner or later and should be reported by kernel as invalid ELF - we've wasted too many hours debugging broken ELFs)
2. There are indeed absolute relocations in test-related ELFs:

closed 2021-09-14 by agkaminski