phoenix-rtos/phoenix-rtos-project #184: kernel: waitpid with WNOHANG option waits for a process state to change instead of returning value at once

damianloew, 2021-09-13:
The problem occurs when creating more than one processes in turn

Test, which reproduce issue: 

https://github.com/phoenix-rtos/phoenix-rtos-tests/pull/49

nalajcie, 2021-09-20:
Yep, I confirm that there is a problem with `waitpid` syscall implementation when `pid != -1` (any process) and there is at least one other pending zombie child. In that case:
 - `WNOHANG` option doesn't work
 - the `waitpid` turns into busy-waiting loop without any sleep inside taking as much CPU as it can

@damianloew with this information You could probably refine the testcase to trigger the error every time (spawn 2 childs - one which will exit soon, one which will never exit, sleep some time for the 1st one to become zombie process, use `waitpid` with `WNOHANG` to wait for the second process).

damianloew, 2021-09-20:
Thanks, I updated test case

agkaminski, 2022-02-11:
Fix here: https://github.com/phoenix-rtos/phoenix-rtos-kernel/pull/302

agkaminski, 2022-02-15:
Fixed in 1cf2a6d86190ef1369f1b7ce471bb330cadb7452

closed 2022-02-15 by agkaminski