phoenix-rtos/phoenix-rtos-project #194: psh: some binaries aren't visible when type ls command

damianloew, 2021-09-24:
For example here 'ls' binary isn't visible in bin. 
[image]

'ls' uses 'readdir' function, which reads subsequent files in chosen directory. It seems like that files aren't visible for 'readdir' also. Mentioned binaries are present in _fs directory in host-pc and can be executed using exec.

The problem occurs, when there is some larger amount of binaries in directory, because when building without test component all executable files are visible.

When creating even several hundreds of files (not executable, just normal files) using touch the problem doesn't occur.

Test with test case reproducing issue (assert_ls_pshcmds: checking if all psh commands are listed in bin):
https://github.com/phoenix-rtos/phoenix-rtos-tests/pull/52

nalajcie, 2021-09-24:
Please specify the `TARGET` as it's probably connected with the filesystem being used for `/bin`. I cannot reproduce the issue when building `ia32-generc` (busybox ls returns the same number of binaries in `/bin` - 85).

damianloew, 2021-09-27:
The problem doesn't occur, when building all components. Please try build using following command:
'TARGET=ia32-generic ./phoenix-rtos-build/build.sh clean core fs test project image'

damianloew, 2021-09-27:
Missing commands in Github Actions runner:
[image]

nalajcie, 2021-09-28:
I can confirm that the issue is present in the disk file produced as an artifact of CI showcasing the problem.

Extracted the ext2 partition and `e2ls` shows the correct `/bin` contents, to do it by Yourself:

```bash
dd if=phoenix-ia32-generic.disk of=part.ext2 skip=4096 count=100000 bs=512
e2ls part.ext2:/bin/
```

I don't know if this is an ext2 issue of `readdir` implementation one (no way to test it on `ia32-generic` because of https://github.com/phoenix-rtos/phoenix-rtos-project/issues/114).

damianloew, 2021-09-28:
It returns correct list with all commands, when I use 
`dd if=phoenix-ia32-generic.disk of=part.ext2 skip=4096 count=100000 bs=512
e2ls part.ext2:/bin/`

damianloew, 2022-03-23:
Seems to work properly now (5f7a0eb3677a9cb7f6e0c6cd76a98be3fee90cbe phoenix-rtos-project). Test case to cover it is present in ls tests: https://github.com/phoenix-rtos/phoenix-rtos-tests/pull/52

closed 2022-03-23 by damianloew