phoenix-rtos/phoenix-rtos-project #235: stm32l4 - uart communication cannot process continuous data fast enough

niewim19, 2021-11-18:
`stm32l4` UART communication cannot process continuous block of data fast enough.

Everything is fine when a real human types communicates with target (e.g with picocom) but using some script that sends data without human-related delays (times between character strikes) breaks the communication and target _misses_ some characters (as they would never arrive via

anglov, 2023-09-25:
@damianloew 
As DMA UART was merged please do the following steps:
1. Enable DMA on default console UART via board_config.h
2. Run tests (CI should do relevant test and all of them should pass)
3. Remove workaround in phoenix-rtos-tests https://github.com/phoenix-rtos/phoenix-rtos-tests/blob/6cd0ac76475196a268db2fe95b6f735c97e6b2b2/trunner/target/armv7m4.py#L104C1-L108C1
4. Run tests.

If everything will pass, then this issue can be closed.

damianloew, 2024-10-01:
Yeah, workaround removed and it works stably - I am closing

closed 2024-10-01 by damianloew