phoenix-rtos/phoenix-rtos-project #252: psh: system crash after `touch` on binaries

niewim19, 2021-12-15:
`touch /bin/psh` or `touch /bin/ls` commands results in system crash. Two screens below to replicate:
1)
[image]
2)
[image]
h

niewim19, 2021-12-16:
Additionally this command completely erases files if they are present before command call.

This is due to use of `fopen()` function with option `"w"`. Libphoenix `open()` lacks support of `O_DIRECTORY` which is posix   -required so this needs bigger fix.

niewim19, 2021-12-16:
- https://github.com/phoenix-rtos/phoenix-rtos-utils/pull/112

Mentioned PR neutralizes `touch` command problems, and makes it _almost_ working if not for the lack of proper libphoenix/phoenix-rtos-kernel stubs and lack of implementations.

damianloew, 2022-12-21:
Resolved

closed 2022-12-21 by damianloew