phoenix-rtos/phoenix-rtos-project #259: `fread()` does not check for `stream == NULL`

niewim19, 2021-12-17:
`fread_unlocked()` shall check if stream pointer is not `NULL` and return `EOF` with `errno = EBADF`.

Instead it crashes the system

agkaminski, 2021-12-17:
IMHO expected behaviour, on linux such mistake causes page fault, not graceful error

niewim19, 2021-12-17:
Can`t argue with that it works like that on linux :)

Documentation states that:
 > The fgetc() function shall fail if data needs to be read and:
 (...)
[EBADF] The file descriptor underlying stream is not a valid file descriptor open for reading. 
 (...)

`NULL` file descriptor is neither valid file descriptor nor valid file descriptor open for reading. So returning `EBADF` seems conformant.

Comes down if we want the little time addition of:
``` c
	if (!stream)
		return SET_ERRNO(-EBADF);
```
in `fread_unlocked()`

niewim19, 2021-12-17:
Unsolvable as only NULL check can be performed explicitly but not for other "wild" pointers not pointing to file. Closed.

closed 2021-12-17 by niewim19