phoenix-rtos/phoenix-rtos-project #263: libphoenix: `fseek()` called on read access file descriptor does not fail

niewim19, 2021-12-17:
According to docs of `fseek()`:
 > The fseek() and fseeko()  functions shall fail if (...):
 [EBADF] The file descriptor underlying the stream file is not open for writing or the stream's buffer needed to be flushed and the file is not open

In libphoenix following code:
```c
fd = fopen(stdpath, "r");
fseek(fd, 0, SEEK_SET);
```
does not produce an error

agkaminski, 2023-06-20:
It doesn't make much sense, why would we forbid fseek on read only stream? It looks like a type (most likely some copy-paste error) on opengroup. There are even examples with read only stream:
https://www.geeksforgeeks.org/fseek-in-c-with-example/

agkaminski, 2023-08-30:
Above example code doesn't produce error on linux either, so I'm closing this issue