phoenix-rtos/phoenix-rtos-project #273: psh: executing psh using `sysexec` and then running `exit` causes an exception

damianloew, 2021-12-22:
Tested on ia32-generic target (qemu). The problem doesn't occur on `armv7m7-imxrt106x`.

[image]

agkaminski, 2021-12-22:
This issue is caused by executing programs from syspage - both sysexec and hm execute from syspage.
This causes object -1 to be passed to exec and consequently to process_load().
process_load() maps data section as NEEDS_COPY, but for object -1 (physical memory) this flag is effectively ignored (by design).

IMHO there are two solutions:
- Never execute object -1 - create memory object before executing from syspage,
- Copy data segment in process_load manually for case of object -1

agkaminski, 2021-12-22:
Dirty fix: Change else if(...) to else here: https://github.com/phoenix-rtos/phoenix-rtos-kernel/blob/4e89fbccfa73e749c7944841f4d4c802010c3e49/vm/map.c#L624

Might break fork() in a very bad way (if e.g. driver has mapped some physical memory of a device)

agkaminski, 2023-06-20:
As the dirty fix has been merged I'm closing this issue as solved

closed 2023-06-20 by agkaminski

damianloew, 2023-06-20:
Yeah, seems to be resolved. Thanks!