phoenix-rtos/phoenix-rtos-project #281: kernel: `vm_pageinfo()` for nommu should signalize lack of page structure

niewim19, 2022-01-04:
Call to `vm_paeinfo()` on nommu target leaves `page` substructure in `_meminfo_t` structure undefined.

https://github.com/phoenix-rtos/phoenix-rtos-kernel/blob/4e89fbccfa73e749c7944841f4d4c802010c3e49/vm/page-nommu.c#L154-L164

This does not give information that data under `map` pointer is invalid.

niewim19, 2022-01-04:
IMHO we can set `page.mapsz` to `-1` or `0` indicating that `map` pointer is invalid/undefined, and/or set `map` to `NULL`

damianloew, 2022-01-04:
This issue causes reading undefined memory by `mem -p` psh command on non-mmu targets, for example `armv7m7-imxrt106x`:
[image]

damianloew, 2022-01-18:
Fixed by https://github.com/phoenix-rtos/phoenix-rtos-kernel/pull/281 and https://github.com/phoenix-rtos/phoenix-rtos-utils/pull/114

closed 2022-01-18 by damianloew