phoenix-rtos/phoenix-rtos-project #285: dummyfs: creates "un`stat()`able" files in syspage

niewim19, 2022-01-10:
# Description 
dummyfs during creation of files in `/syspage/` directory gives them posix incompatible parameters. During `stat()` call on such entries it returnes whatever it finds in value `0xaBadBabe` that is (by default) stored in `dummyfs_entry->mode` field.

# Trace
Permalink to  macro `OBJECT_MODE_MEM` with value `0xBadBabe` definition: 
https://github.com/phoenix-rtos/phoenix-rtos-filesystems/blob/42e344cbdf7493ad0f5d8a71a486aed65ccc7f71/dummyfs/dummyfs.c#L35

Permalink to `dummyfs_createMapped()` which writes `OBJECT_MODE_MEM` into `mode` field:
https://github.com/phoenix-rtos/phoenix-rtos-filesystems/blob/42e344cbdf7493ad0f5d8a71a486aed65ccc7f71/dummyfs/dummyfs.c#L782

Permalink to syspage-related usage of `dummyfs_createMapped()`:
https://github.com/phoenix-rtos/phoenix-rtos-filesystems/blob/42e344cbdf7493ad0f5d8a71a486aed65ccc7f71/dummyfs/srv.c#L52

niewim19, 2022-01-10:
@agkaminski can you give it a look?

niewim19, 2022-01-14:
As this causes issues on some targets where `exec()` works but we use dummyfs and syspage, proposed solution is to substitute `0xaBadBabe` magic number with `0x81FD` which should correspond to normal file with execution privilages

agkaminski, 2022-01-17:
Changing this mode is not enough - we need to handle this one too: https://github.com/phoenix-rtos/phoenix-rtos-filesystems/blob/0d1f09378ff390a6308ebba052821529ba0bd137/dummyfs/dummyfs.c#L498

So some kind of flag is needed here to be able to handle this correctly (either additional variable or reverved bit in mode).

Also, it would be nice to type this "0x81FD" as a combination of flags, not a magic number.

niewim19, 2022-01-18:
Solution merged to phoenix-rtos-filesystems. Issue is closed.

closed 2022-01-18 by niewim19