phoenix-rtos/phoenix-rtos-project #295: armv7m7-imxrt106x: reboot after loading test-echo (using app) and calling go! command

damianloew, 2022-01-18:
- Happens once every three/four times
- Verified when using test runner and when running manually
- For now verified on an only one test unit
- Does not happen before last submodules update (c0c97d2926d394314d9616e87b93ef7dd36736ae)
- Output from test runner:

[image]

damianloew, 2022-01-18:
UPDATE 1:
- undoing following commits causes no error (for example 3/10 launches with these changes causes the error, without: 0/20):
  - Add 'atBlocks' and 'atIOBlock' attributes support (phoenix-rtos-filesystems)
  - posix/fstat: add 'atBlocks' and 'atIOBlock' attributes for setting stat.st_blocks and stat.st_blksize fields (phoenix-rtos-kernel)
  - sys/stat: add 'atBlocks' and 'atIOBlock' attributes for setting stat.st_blocks and stat.st_blksize fields (libphoenix)
- changing test-echo.c content to simple printf fixes problem (but I've verified that test-echo doesn't start before psh, not seen in ps output, when 20s sleep is applied)
- Seems that test-echo binary is placed somewhere (app command), where can be executed by accident in psh start (?)

damianloew, 2022-01-19:
UPDATE 2:

- with changed test-echo.c content as follows 1/50 launches fails with following exception (other than before)
[image]
[image]

Maxez, 2022-02-04:
The problem is caused by `main_initthr()` kernel stack overflow. I've checked its `kstack` under debugger (it should contain `0xba` pattern set in `proc_threadCreate()` ([link](https://github.com/phoenix-rtos/phoenix-rtos-kernel/blob/71c7195527b244648c621c55e3127cb2cb05b62d/proc/threads.c#L784)) but got overwritten.

closed 2022-02-15 by agkaminski