phoenix-rtos/phoenix-rtos-project #300: libphoenix: `strtod()` does not support hex convertion + no errno handling

niewim19, 2022-01-26:
`strtod()` is described in C89 out implementation should conform to its documentation. According to docs it should support hexadecimal conversion. Our implementation lacks it.

Additionally out implementation does not handle errno at all and it should in at least one case:
 > These functions shall fail if:
[ERANGE] The value to be returned would cause overflow or underflow.

HBuczynski, 2022-02-11:
Commit message in `phoenix-rtos-project` does not reference to this issue. The same PR identity in `phoenix-rtos-kernel` as this issue id.

Darchiv, 2024-07-05:
`errno` issue fixed in https://github.com/phoenix-rtos/libphoenix/pull/193

Closing, as there is now a more specific issue for hex values: https://github.com/phoenix-rtos/phoenix-rtos-project/issues/703

closed 2024-07-05 by Darchiv