phoenix-rtos/phoenix-rtos-project #397: dummyfs: readdir() returns EINVAL, when entry name length equals NAME_MAX

damianloew, 2022-04-28:


damianloew, 2022-04-29:
I agree that it should be changed in https://github.com/phoenix-rtos/libphoenix/blob/8ab382d5b4007ebad85f1df901b9f8f379c1183e/unistd/dir.c#L347, but it doesn't solve the problem. I verified that in this case `EINVAL` is return
ed, like in the screenshot above. I will opt for waiting for @Maxez, but maybe increasing `msg.o.size`, like below:
[image]

Maxez, 2022-05-04:
I agree with @nalajcie - it's a bug in `readdir` libphoenix implementation.
We need to allocate `sizeof(struct dirent) + NAME_MAX + 1` bytes for `s->dirent` and correctly pass it in `msg.o.data` and `msg.o.size` fields.

damianloew, 2022-05-05:
> I agree with @nalajcie - it's a bug in `readdir` libphoenix implementation. We need to allocate `sizeof(struct dirent) + NAME_MAX + 1` bytes for `s->dirent` and correctly pass it in `msg.o.data` and `msg.o.size` fields.

Thanks, created PR with the mentioned changes. I would be grateful for a review.

closed 2022-05-06 by damianloew