phoenix-rtos/phoenix-rtos-project #400: libphoenix: `printf()` cannot print floats bigger than 2^20

niewim19, 2022-05-06:
# Test scenario:
Printing floating point values larger than 2^20 results in wrong number printed

# Test case:
Test code:
```c
	printf("%f\n", 1048575.F);   /* 2^20 - 1 */
	printf("%f\n", 1048576.F);   /* 2^20 */
	printf("%f\n\n", 1048577.F); /* 2^20 + 1 */

	printf("%f\n", 1048575.); /* 2^20 - 1 */
	printf("%f\n", 1048576.); /* 2^20 */
	printf("%f\n", 1048577.); /* 2^20 + 1 */
```
results in:
```
1048575.000000
1048576.000000
1099511627777.000000

1048575.000000
1048576.000000
1099511627777.000000
```

niewim19, 2022-05-06:
Issue forces changes in tests proposed in 
 - https://github.com/phoenix-rtos/phoenix-rtos-tests/pull/75

gerard5, 2023-02-20:
Mention of pull-request which extends support for `%f` (float, doubles) -> https://github.com/phoenix-rtos/libphoenix/pull/228

astalke, 2023-03-13:
Fixed in: 
- https://github.com/phoenix-rtos/libphoenix/pull/228

closed 2023-03-13 by astalke