phoenix-rtos/phoenix-rtos-project #409: libphoenix: `rename("f1", "f2")` fails if `f2` exists (posix incompatibility)

damianloew, 2022-05-26:
to reproduce:

 - place the following code in `_user/hello/main.c`
[image]
- run hello binary
- the result (EEXISTS errno):
[image]

According to posix it should remove `testfile1` and then rename `testfile2` to `testfile1`.

Detected in mbedtls tests.

damianloew, 2022-05-26:
Proposed fix in https://github.com/phoenix-rtos/libphoenix/pull/180

nalajcie, 2022-05-26:
Please note that POSIX requires `rename` operation to be atomic, relevant citation:

>  If the link named by the new argument exists, it shall be removed and old renamed to new. In this case, **a link named new shall remain visible to other threads throughout the renaming operation** and refer either to the file referred to by new or old before the operation began.

This means that the `new` should always be resolvable so implementation with `unlink` + `link` is non-conformant.