phoenix-rtos/phoenix-rtos-project #411: imx6ull-uart: `getline()` returns randomly chopped data

niewim19, 2022-06-02:
# Issue sumary
Using `getline()` on imx6ull-uart driver returns data randomly chopped with no respect to real `\n` that appear in input data.

# Test scenario
 1. As an example a GPS module is used that communicates via UART (baud 9600, 8N1). It is connected to UART2 on imx6ull evalkit.
 2. Second uart driver is spawned using psh script (see _Appendix 1_)
 3. Test program (see _Appendix 2_) is run that calls `fopen()` on uart in `/dev/`, and calls `getline()` ten times, each time printing what was read.

# Test output
Data output from test program:
```
-> $GPGGA,122558.000,5223.5
-> 16,47,217,50,26,22,195,45,08,,,30*49
$GPRMC,12255
-> 55.5533,E,1,3,76.72,111.3,M,38.7,M,,*62
$GPGSA,A,2,16,10,
-> 0.08,K,A*30
$GPGGA,122600.000,522
-> 1,N,02055.5532,E,0.08,0.00,020622,,,A*68
$GPVTG,0.0
-> ,4.72,4.61,1.00*02
$GPRMC,
-> N,02055.5532,E,1,3,4.61,111.3,M,38.7,M,,*5B
$GPGSA,A,2,16,10,2
-> ,K,A*32
$GPGGA,122603.000,522
-> 6,47,217,50,26,22,195,46,08,,,29*42
$GPRMC,122603.0
-> 5529,E,1,3,4.61,111.3,M,38.7,M,,*5F
$GPGSA,A,2,16,10,2
```

Data arriving on UART (acquired via `read()` with big buffer, not the same data as in previous output):
```
$GPRMC,235953.800,V,,,,,0.00,0.00,050180,,,N*42
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPRMC,235954.776,V,,,,,0.00,0.00,050180,,,N*4B
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,235955.776,,,,,0,0,,,M,,M,,*43
```

# Appendix 1: psh script for uart driver start
``` bash
:{}:
X /syspage/imx6ull-uart 0 2 9600 0 0
```

# Appendix 2: test program
``` c
#include <stdio.h>
#include <unistd.h>
#include <fcntl.h>

int main(int argc, char ** argv)
{
    char * buf = NULL;
    size_t sz;
    int i = 0;

    FILE *fd = fopen("/dev/uart2", "r");
    if (fd != NULL) {
        for (i = 0; i < 10; i++) {
            sz = getline(&buf, &sz, fd);
            printf("-> %s\n", buf);
        }
        fclose(fd);
    }
    return 0;
}
```

niewim19, 2022-06-02:
Thank you for explanation.

And yeah - `fd` brings confusion - thanks for that :)

gerard5, 2023-02-20:
The `getline()` -> `getdelim()` function has been completely rewritten (https://github.com/phoenix-rtos/libphoenix/commit/4d879ee4baef7f3270782651b5bf2104dbb04525). I am interested in whether the topic is still valid, and if not, whether the correct operation can be confirmed and the topic closed.

nalajcie, 2023-06-20:
the `getline`/`getdelim` got another rewrite, can you check if the issue is still present? Maybe @damianloew can help you with testing?

damianloew, 2023-06-20:
okay, we will check it with @maska989

maska989, 2023-06-21:
[image]

Tested on imx6ull with jffs2 onboard. @gerard5 @nalajcie
I think job is done and everything looks fine

nalajcie, 2023-06-21:
great, maybe we could incorporate this real-world data in unit tests to avoid regressions in the future and close the issue?

closed 2023-08-30 by agkaminski