phoenix-rtos/phoenix-rtos-project #424: armv7a7-imx6ull-evk: can't communicate with psh

damianloew, 2022-06-13:
For the latest commit in phoenix-rtos-project (d464da30ea18b1c79764fa841f91a23b261ea070). When uploading image to ram as required (https://github.com/phoenix-rtos/phoenix-rtos-doc/blob/master/quickstart/armv7a7-imx6ull.md) it seems like uart read does not work. There are no characters printed when entering them on the keyboard. Buliding with the following command: `TARGET=armv7a7-imx6ull-evk ./phoenix-rtos-build/build.sh clean core fs image project`
[image]

damianloew, 2022-06-13:
Probably the latest commit that works fine: de6a0627fcfd6d8229d557513e1ef7c9d7c0b9e1

damianloew, 2022-06-13:
The commit in `phoenix-rtos-devices` that causes the issue: https://github.com/phoenix-rtos/phoenix-rtos-devices/commit/ba3a267df9f35a3161f4e4e2ca4e9990c198ebea

damianloew, 2022-06-13:
It's the following change (not sure why it was applied):
[image]

nalajcie, 2022-06-14:
now RX doesn't work if the device isn't open (which is the correct behavior and saves some cpu clocks). Please verify if `psh` correctly opens the tty device file (maybe it's a problem with `psh` input params)

damianloew, 2022-06-21:
After setting this macro console works, but very 'slowly
[image]
'

damianloew, 2022-12-22:
It seems to no longer occur, when using b64d8e617b5e68b6df4d6781a8ed1328a5ed1783 phoenix-rtos-project revision.

closed 2022-12-22 by damianloew