phoenix-rtos/phoenix-rtos-project #457: Problem with micropython format()

PNieck, 2022-07-13:
Expression
```python
'{0:0}'.format('zzz')
```
throws `ValueError` exception, but it shouldn't.

PNieck, 2022-07-13:
Problem was found during this test:
```python
# tests for errors in {} format string

try:
    '{0:0}'.format('zzz')
except (ValueError):
    print('ValueError')

try:
    '{1:}'.format(1)
except IndexError:
    print('IndexError')

try:
    '}'.format('zzzz')
except ValueError:
    print('ValueError')

# end of format parsing conversion specifier
try:
    '{!'.format('a')
except ValueError:
    print('ValueError')

# unknown conversion specifier
try:
    'abc{!d}'.format('1')
except ValueError:
    print('ValueError')

try:
    '{abc'.format('zzzz')
except ValueError:
    print('ValueError')

# expected ':' after specifier
try:
    '{!s :}'.format(2)
except ValueError:
    print('ValueError')

try:
    '{}{0}'.format(1, 2)
except ValueError:
    print('ValueError')

try:
    '{1:}'.format(1)
except IndexError:
    print('IndexError')

try:
    '{ 0 :*^10}'.format(12)
except KeyError:
    print('KeyError')

try:
    '{0}{}'.format(1)
except ValueError:
    print('ValueError')

try:
    '{}{}'.format(1)
except IndexError:
    print('IndexError')

try:
    '{0:+s}'.format('1')
except ValueError:
    print('ValueError')

try:
    '{0:+c}'.format(1)
except ValueError:
    print('ValueError')

try:
    '{0:s}'.format(1)
except ValueError:
    print('ValueError')

try:
    '{:*"1"}'.format('zz')
except ValueError:
    print('ValueError')

# unknown format code for str arg
try:
    '{:X}'.format('zz')
except ValueError:
    print('ValueError')
```
Other examples works in a correct way.

nalajcie, 2022-07-13:
IMHO this is ok, host `micropython` behaves the same way - so it's not an issue with our RTOS
[image]

Also, CPython up to 3.9 behaved the same, the `format` mini-language was improved in 3.10 (see related issue: https://bugs.python.org/issue27772)

Please note that micropython implements mostly python 3.4 with some of the newer features but it's not 100% compatible (see http://docs.micropython.org/en/latest/genrst/index.html for main differences).

Closing this as invalid, feel free to reopen if micropython on RTOS behaves differently than on host OS.

closed 2022-07-13 by nalajcie