phoenix-rtos/phoenix-rtos-project #545: plo: new error after last changes: `Can't open file on usb0`

damianloew, 2022-11-07:
- after the following changes in plo `can't open` error may occur in ~1/8 test runner campaigns. Even if it was not seen before, there could be different reason.
    - 152a cmd: apply changes in interrupts API
    - bdc9 devices: apply changes in interrupts API
    - f6a0 hal/armv7m: allow to disable specific interrupt in NVIC
    - 7718 hal/ia32: apply changes in interrupts hal API
    - 0958 hal/armv7a: allow to disable specific interrupt in GIC
 
 - below is a test runner fail with output from phoenixd:
 
[image]

 - after error plo seems to work properly (i. e. sending next commands)
 
 - caught when testing https://github.com/phoenix-rtos/phoenix-rtos-project/commit/1d61317bd06779bc2bcfb08f328f99f276842d76 phoenix-rtos-project with only plo changes (not caught when testing with commits to `a6df zynq7000-flash: add erase handler implementation`)

 - couldn't reproduce yet when using the current phoenix-rtos-project revision: https://github.com/phoenix-rtos/phoenix-rtos-project/commit/dc5d14c4f82c7b766546967c4603385f6886f06f

agkaminski, 2022-11-07:
I don't really see anything dangerous in these commits, maybe it somehow made some other problem more apparent?

damianloew, 2022-11-07:
I think so, please let me know what exactly can I check. Maybe it's related to https://github.com/phoenix-rtos/phoenix-rtos-project/issues/489

agkaminski, 2022-11-07:
My thought for future debugging - it might be worth checking if `hal_interruptsDisable` really do disable interrupt

damianloew, 2023-02-17:
It seemed to occur only on damaged imxrt117x-unit, so I think it's probably not a problem with system.

damianloew, 2023-06-16:
As it was discovered during nightly run (https://github.com/phoenix-rtos/phoenix-rtos-project/actions/runs/5285224467/jobs/9563608911) this issue may still happen, even on a board without any fixes. This happen for the following phrtos-project revision: https://github.com/phoenix-rtos/phoenix-rtos-project/commit/ab93a3aeac748f002075fb997f1290f87b6bd19e

damianloew, 2023-06-16:
Probably related with: https://github.com/phoenix-rtos/phoenix-rtos-project/issues/710

damianloew, 2023-10-09:
Update: the very similar issue is still being encountered, both on imxrt106x and imxrt117x. There are usually 2 following error messages:
[image]

[image]

There is nothing suspicious in dmesg output during those fails. It seems that phoenixd is launched too early and `0.5s` sleep after finding an ACM device in test runner solves the problem.

damianloew, 2024-10-01:
Not seen anymore - probably fixed, I am closing it for now

closed 2024-10-01 by damianloew