phoenix-rtos/phoenix-rtos-project #552: psh: ls -l: year is not printed for files with non current year dates

damianloew, 2022-11-18:
Current behavior:
[image]
Intended behavior:
[image]

jmaksymowicz, 2023-09-25:
I haven't noticed the problem with seconds being printed instead of year. The current version of `ls -l` command never prints the year. The original title of this issue is correct though, other implementations of `ls` do this so it would be useful to have this feature.

nalajcie, 2023-09-25:
Haven't tested the current code but until recently this still was a problem (DTR project 1.07.04, jffs2 rootfs):

[image]

damianloew, 2023-09-25:
> I haven't noticed the problem with seconds being printed instead of year. The current version of `ls -l` command never prints the year. The original title of this issue is correct though, other implementations of `ls` do this so it would be useful to have this feature.

Yeah, that's the point - it would be useful to print year if it's different than the current one.