phoenix-rtos/phoenix-rtos-project #576: Virtual address issue

KaduruSreedhar, 2022-12-15:
**Reference :** 
Sifive_fu board DDR physical address is (0x8000_0000-0x1F_7FFF_FFFF)
PLIC address is 0x0C00_0000, 
VADDR_KERNEL_INIT=0x0000_003F_C000_0000
plic_common.regw = (void *)((u64)((1L << 39) - 1024 * 1024 * 1024 + 0x0c000000)| (u64)0xffffff8000000000);
static void *dtb_relocate(void *addr)
{
	return (void *)((u64)((1L << 39) - (u64)2 * 1024 * 1024 * 1024 + addr - 0x80000000L) | (u64)0xffffff8000000000);
}

**Now I am using below configuration RISCV board :** 
DDR physical address is (0x4000_0000-0x7FFF_FFFF)
PLIC address is 0x0100_0000

**Please  suggest :** 
VADDR_KERNEL_INIT=?
plic_common.regw =?
static void *dtb_relocate(void *addr)
{
	return (void *)(?)
}

agkaminski, 2022-12-15:
Hi,

right now we only support spike and qemu RISC V targets, but I agree that this value should not be hardcoded in this place. Will change that on the first occasion.

My suggestion (not tested):
VADDR_KERNEL_INIT should stay as is (as it's not platform dependent in any way, this is a virtual address),
plic_common.regw = (void *)((u64)((1L << 39) - 1024 * 1024 * 1024 + 0x01000000)| (u64)0xffffff8000000000);
static void *dtb_relocate(void *addr)
{
return (void *)((u64)((1L << 39) - (u64)2 * 1024 * 1024 * 1024 + addr - 0x40000000L) | (u64)0xffffff8000000000);
}

There is a good chance that RAM physical address need change in other places too, e.g.:
https://github.com/phoenix-rtos/phoenix-rtos-kernel/blob/a534065b8ed631809e637b55f5a67c1417dcdb1c/hal/riscv64/pmap.c#L270
https://github.com/phoenix-rtos/phoenix-rtos-kernel/blob/a534065b8ed631809e637b55f5a67c1417dcdb1c/hal/riscv64/pmap.c#L483

KaduruSreedhar, 2022-12-15:
Thanks for your input, it worked. But nothing seems to be happening after proc_start.
Please refer the below images.

[image]
[image]
[image]
[image]

KaduruSreedhar, 2022-12-15:
We ran test_proc_threads and got the following results.
n is reduced to 10000

[image]

[image]

agkaminski, 2022-12-15:
It seems to me that something might still be wrong with either interrupts or interrupts controller - system is crashing right after enabling interrupts

closed 2026-01-21 by agkaminski