phoenix-rtos/phoenix-rtos-project #580: armv7m7-imxrt117x-evk: BusFault when starting new psh after exit cmd

damianloew, 2023-01-02:
[image]

jsarzy, 2023-01-12:
Additionaly system can hang after executing ps command (there is no output). It seems that it happen only on the imxrt117x target and the imxrt106x is not affected.

gerard5, 2023-01-12:
FYI: Probably the source of problems is toolchain. Just cloned the phoenix-rtos-project sha:e1224d, manually build toolchains (I keep an archive), here's the quick track:
* Using toolchain build on 24.09.2021 (sha:d275ff) this does not occur see the attached picocom logs [20230112_230433_ttyACM0.log](https://github.com/phoenix-rtos/phoenix-rtos-project/files/10406458/20230112_230433_ttyACM0.log)
* When used (my current) toolchain build on 18.11.2022  (sha:efdff3) above mentioned failure occurs to me as well.

Hopefully this helped.

damianloew, 2023-09-11:
> can we confirm if this still happens?
> 
> I think since the issue detection there were some changes to TLS handling on Cortex-M targets in `libphoenix` but I'm not sure.

Seems to be fine now, tested using 8970699c419155287f806f14d5ff7166df4a0d7f phrtosproject version.

nalajcie, 2023-09-12:
So, can we remove this: 

https://github.com/phoenix-rtos/phoenix-rtos-tests/blob/fd0b9a58935a48865ab594959a0e60fcd3fd0b71/trunner/target/armv7m7.py#L104-L106

nalajcie, 2023-09-14:
The issue is still present, there is no info about exception but consecutive:

```
sysexec psh
exit
sysexec psh
exit 
sysexec psh
exit
```

would hang on `sysexec psh` at some point along the way

damianloew, 2024-06-10:
Discovered another time on ea760a2fdc7ebb328303851bb29f086b3f4775c0

anglov, 2024-06-10:
Could you put exception log?

damianloew, 2024-06-10:
> Could you put exception log?

Sorry, it was checked on the older version of toolchain, I am still testing it to be 100% sure, but an issue seems to be resolved

damianloew, 2024-06-10:
The issue no longer occur, checked on ea760a2fdc7ebb328303851bb29f086b3f4775c0

Probably fixed in https://github.com/phoenix-rtos/phoenix-rtos-build/commit/072b8599ebcf4178d0094d3ae256dd4e2eddba0c

closed 2024-06-10 by damianloew