phoenix-rtos/phoenix-rtos-project #596: gh-actions: remaining git locks after cancelling jobs

damianloew, 2023-02-02:
- When a job is cancelled, for example because of other job's fail or force-push it may happen that executing some git command is interrupted. Like there:
 
[image]

 https://github.com/phoenix-rtos/phoenix-rtos-corelibs/actions/runs/4015577123/jobs/6897506309

 - In the result, lock git files, like shallow.lock remain in the working repository

 - actions/checkout uses the existing files, when running self-hosted runner (run tests hw job)

- Then we get the `Another git process seems to be running in this repository` error:

[image]

https://github.com/phoenix-rtos/phoenix-rtos-corelibs/actions/runs/4015605831/jobs/6897575569

nalajcie, 2023-02-02:
This seems like an error in  https://github.com/actions/checkout/ action - from reading the code it seems that the locks are being cleared in the main repo dir (if exists) but not in the submodules' dirs.

Probably we should create an issue in upstream repo (or try to fix it ourselfs + prepare pull request, but I'm not sure we have people fluent in TypeScript).

damianloew, 2023-02-02:
> This seems like an error in https://github.com/actions/checkout/ action - from reading the code it seems that the locks are being cleared in the main repo dir (if exists) but not in the submodules' [...]

Thanks for checking it. I've found the following issue, that sounds similar:
https://github.com/actions/checkout/issues/598

Alternatively I see the following temporary solution in our workflows (not tested yet):

[image]

nalajcie, 2023-02-02:
I would prefer this to be done as a part of git action so either fixed in upstream or we should fork and fix the issue ouselfs.

The issue linked is not related - it treats about file locking on windows runners.