phoenix-rtos/phoenix-rtos-project #599: dummyfs: extended area isn't always zero-filled after truncate/ftruncate

damianloew, 2023-02-08:
According to POSIX mentioned area should always be zero-filled after truncating up:

truncate:
[image]

ftruncate:
[image]

 - How to reproduce:
 
  - Run the following program on dummyfs target until it fails (should be visible after ~3 runs)
  ```C
#include <stdio.h>
#include <fcntl.h>
#include <sys/stat.h>
#include <unistd.h>
#include <string.h>

#define LINE1 "line1"
#define FNAME "testfile"

static int fd;
static char buf[50];

int setup(void)
{
	/* open a file */
	fd = open(FNAME, O_RDWR | O_CREAT | O_TRUNC, S_IRUSR | S_IWUSR);
	if (fd < 0) {
		fprintf(stderr, "File hasn't been opened correctly\n");
		return -1;
	}
	memset(buf, '\0', sizeof(buf));

	return 0;
}

int teardown(void)
{
	memset(buf, '\0', sizeof(buf));
	if (fd >= 0) {
		if (close(fd) != 0) {
			fprintf(stderr, "File has't been closed correctly\n");
			return -1;
		}
	}

	if (remove(FNAME) != 0) {
		fprintf(stderr, "File has't been removed correctly\n");
		return -1;
	}
}

int tcase(void)
{
	struct stat st;
	int datalen = 2 * sizeof(LINE1);
	int ret;

	/* alter buf with arbitrary data, because '\0' are going to be stored */
	memset(buf, ' ', sizeof(buf));

	/* write LINE1 to the file */
	if ((ret = write(fd, LINE1, sizeof(LINE1))) != sizeof(LINE1)) {
		fprintf(stderr, "All bytes haven't been writte, ret = %d\n", ret);
		return -1;
	}

	if (close(fd) != 0) {
		fprintf(stderr, "File has't been closed correctly\n");
		return -1;
	}

	/* truncate file length to higher than it already is */
	if (truncate(FNAME, datalen) != 0) {
		fprintf(stderr, "Truncate failed\n");
		return -1;
	}

	fd = open(FNAME, O_RDONLY, S_IRUSR);

	if (fd < 0) {
		fprintf(stderr, "File hasn't been opened correctly\n");
		return -1;
	}

	if ((ret = read(fd, buf, datalen)) != datalen) {
		fprintf(stderr, "All bytes haven't been read, ret = %d\n", ret);
		return -1;
	}

	for (int i = 0; i < datalen; i++)
	{
		printf("file[%d] = %c (%d)\n", i, buf[i], buf[i]);
	}

	return 0;
}

int main(void)
{
	if (setup() < 0) {
		return 1;
	}
	if (tcase() < 0) {
		return 1;
	}
	if (teardown() < 0) {
		return 1;
	}
}

  ```

   - result (some junk output visible):
   
[image]

badochov, 2023-02-21:
I've done a PR solving this however in my testing on imx6ull I wasn't able to observe this bug even before the fix. https://github.com/phoenix-rtos/phoenix-rtos-filesystems/pull/91

closed 2023-02-21 by badochov

damianloew, 2023-02-21:
Thanks! I confirm that the above issue does no longer occur on imxrt106x target (I assume on other dummyfs targets too)