phoenix-rtos/phoenix-rtos-project #604: Fscanf floating point problems

maska989, 2023-02-14:
- **All errors occur when floating points value are used in **fscanf/sscanf/vfscanf/vsscanf** on ia32-generic-qemu**

 - To reproduce issue copy following code:
```c
#include <stdio.h>
#include <string.h>
#include <stdint.h>
#include <stddef.h>
#include <stdlib.h>
#include <stdarg.h>
#include <assert.h>
#include <fcntl.h>

int main(void)
{
	
	char value[] ="123.123";
	float value_copy;
	static FILE *src;
	src = fopen("test","w+");
	fwrite("123.123",1,sizeof(value),src);
	fputc('\n', src);
	fflush(src);
	rewind(src);
	fscanf(src,"%f",&value_copy);
	printf("\nValue of fscanf %f\n",value_copy);
	fclose(src);

}
```
  - Tested on ia32-generic-qemu
[image]

gerard5, 2023-02-17:
@maska989 The issue is `open` from the beginning, it was not a `draft` issue, a solution was created for it, so why was the title of the problem that concerned a specific thing changed to the generalized form instead of creating another separate issue addressing the next problem  ?

maska989, 2023-02-17:
@gerard5 Didn't meant to do any kind of problem i will separate this in different issues and this one will back to original form. Once again sorry for disruption.

gerard5, 2023-02-17:
@maska989 please verify if the provided solution https://github.com/phoenix-rtos/libphoenix/pull/230 resolves the problem addressed by the current issue topic. From my side it looks like your test after applying the aforementioned PR successfully passes the test provided by you. I compared the result with my host machine:

on ia32-generic (qemu):
[image]

on host:
[image]

gerard5, 2023-02-18:
All fixes for this issue have been merged into master. (https://github.com/phoenix-rtos/phoenix-rtos-project/commit/1db5b2d284ddab504c82f06953b0ae2984aeb4cc):
  > stdio/scanf: Add `%[aefgAEG]` conversions
  > stdio/scanf: Fix `%z` length modifier
  > stdlib/strtod: Fix handling of negative values

closed 2023-02-18 by gerard5