phoenix-rtos/phoenix-rtos-project #607: armv7a9-zynq7000: strtod does not handle negative values

damianloew, 2023-02-15:
- Tested on armv7a9-zynq7000-zedboard
 - Seems to work properly on ia32-generic-qemu architecture
 - How to reproduce:
   - Run the following code on mentioned architecture:
   ```C
  #include <stdio.h>
  #include <stdlib.h>
  
  
  int main(void)
  {
	  printf("strtod(-0.008598) = %lf\n", strtod("-0.008598", NULL));
	  printf("strtod(-0.5) = %lf\n", strtod("-0.5", NULL));
	  printf("strtod(-1.0) = %lf\n", strtod("-1.0", NULL));
	  printf("strtod(-2.99) = %lf\n", strtod("-2.99", NULL));
	  return 0;
  }
   ```

   - Result:
   
    [image]

niewim19, 2023-02-15:
Recently there was change in strtof/d/ld implementation in libphoenix:
 - https://github.com/phoenix-rtos/libphoenix/commit/6b37553265786bb55a85da93d46a627b25749ad0, merged in January

This issue does not occur before this commit. Significant change is that `long double` is now used to hold conversion value and `double` was used before.

nalajcie, 2023-02-16:
@DawidSzpejna it might be worth to add these tests to https://github.com/phoenix-rtos/phoenix-rtos-tests/pull/135 (actually I see that this test shows error on zynq already)

closed 2023-02-17 by gerard5