phoenix-rtos/phoenix-rtos-project #609: *Scanf and floating points related problems

maska989, 2023-02-17:
- **All errors occur when floating points value are used in **fscanf/sscanf/vfscanf/vsscanf** on ia32-generic-qemu**

 - To reproduce issue copy following code:
```c
#include <stdio.h>

int main(void)
{
	float val_f;
	float val_g;
	float val_G;
	float val_e;
	float val_E;
	float val_a;
	float val_A;

	float f = 123.123;
	char str_f[]= "123.123";
	sscanf(str_f,"%f",&val_f);
	sscanf(str_f,"%g",&val_g);
	sscanf(str_f,"%G",&val_G);
	sscanf(str_f,"%e",&val_e);
	sscanf(str_f,"%E",&val_E);
	sscanf(str_f,"%a",&val_a);
	sscanf(str_f,"%A",&val_A);

	printf("\n %f // %g // %G // %e // %E // %a // %A\n"
	,val_f, val_g, val_G, val_e, val_E, val_a, val_A);
}

```
  - Tested on ia32-generic-qemu
 
[image]

It occure on : fscanf/sscanf/vfscanf/vsscanf funtions;

gerard5, 2023-02-17:
That is not the case of `*scanf` (after applying my PR from the issue https://github.com/phoenix-rtos/phoenix-rtos-project/issues/604) which adds support for `%f` and other `aefgAEFG`, it scanfs the string correctly. Try changing printf (not supported `%` formats `aeAEG` into eg. `%f` or `%g` verify with for example like that that:
[image]

the output will be:
[image]

maska989, 2023-02-20:
@gerard5 Sorry again for bother. After your advice works fine. Thanks again.

damianloew, 2023-02-21:
As mentioned above, the reason of this issue is wrong printf formatting, so it can be closed

closed 2023-02-21 by damianloew