phoenix-rtos/phoenix-rtos-project #611: plo: MMIO access without going through HAL/drivers

szsam, 2023-02-20:
I have noticed that some code accesses memory-mapped I/O (MMIO) without going through Hardware Abstraction Layer (HAL) or drivers. This type of access may have several drawbacks, including poor portability and reusability.

I would like to open this issue to investigate why some MMIO access does not follow the conventional path through HAL or drivers. Are there specific scenarios where non-conventional MMIO access is preferred or a must (e.g. because, for example, SDK does not provide the necessary API)? Is it possible to fix the code by accessing MMIO conventionally (i.e. through HAL/drivers)?

Code snippets with MMIO access that does not go through HAL or drivers:
https://github.com/phoenix-rtos/plo/blob/c5188460ff8fe95c1f00d79fd36f90d2d474b7c6/devices/disk-bios/disk-bios.c#L52-L53

agkaminski, 2023-02-20:
Hi,

the snippet you provided is a part of a device driver (drivers are placed in `devices` directory) - so it's not supposed to be portable (as it is written specifically for the ia32 target).

szsam, 2023-02-24:
Thanks for your explanation.

closed 2023-02-24 by szsam