phoenix-rtos/phoenix-rtos-project #612: sensorhub: possible off-by-one on nul-terminating messages in `pa6h` driver

niewim19, 2023-02-20:
In `pa6h` sensorhub driver it is possible that nul-terminating messages at the end of `pa6h_receiver()`:

https://github.com/phoenix-rtos/phoenix-rtos-devices/blob/acb0abfed72acc2611e6a69a41efb20a5fdac34e/sensors/gps/pa6h.c#L186-L188

 - The code above sets nul character at the `sz` element of message which is calculated as token pointer difference + 4. It should place nul character one sign earlier. If buffer ends exactly at the `tokenEnd + 3` this means writing outside of buffer.
 - In a rare error event it could even replace `$` of the next message with nul character. It should be checked earlier in code

I am self assigning this issue to not lose it.

niewim19, 2023-02-20:
@kemonats Is this the _outside of buffer write_ issue you have mentioned in one of closed PR-s?

kemonats, 2023-02-21:
Yes.

niewim19, 2023-02-22:
Solved via: https://github.com/phoenix-rtos/phoenix-rtos-devices/pull/343

closed 2023-02-22 by niewim19