phoenix-rtos/phoenix-rtos-project #617: ptrdiff_t length modifier for *scanf problem

maska989, 2023-02-22:
During scanf tests there was a problem reading data assigned as ptrdiff_t ("%t*") .

- In testing fallowing branch was used [https://github.com/phoenix-rtos/libphoenix/tree/astalke/RTOS-356](url) 

- To reproduce issue copy following code:
```c
#include <stdio.h>
#include <stddef.h>
#include <string.h>	
#include <limits.h>

#include <stdint.h>

int main(void)
{
	char str[125];

	sprintf(str,"%td",PTRDIFF_MAX);

 	ptrdiff_t val;
 	ptrdiff_t val1;
	
	printf("\n td = %s",str);

	sscanf(str,"%td",&val);
	printf("\nval with *td = %td",val);

	sscanf("2147483647","%td",&val1);
	printf("\nfinger val with *td = %td",val1);
	
	sscanf("2147483647","%ld",&val1);
	printf("\nfinger val with *ld = %ld",val1);
}
```

Outcome on Host-generic-pc  and ia32-generic-qemu:

[image]

gerard5, 2023-02-22:
@maska989 the above pull-request should resolve this issue, please verify before merge:
[image]

maska989, 2023-02-23:
@gerard5 After applying your pull-request everything works fine. Thanks.

closed 2023-02-23 by gerard5