phoenix-rtos/phoenix-rtos-project #628: Functions `getline` and `getdelim` does not add terminating NUL, when end-of-file is encountered

PNieck, 2023-03-03:
# Example

Suppose that we have such a file without new line character at the end of last line:
```
       ipsum
lorem
```

Then such a program:
```c
int main()
{
	char path[] = "path/to/file";

	FILE *file;
	char *line = NULL;
	size_t lineLen = 0;

	file = fopen(path, "r");
	if (file == NULL) {
		fprintf(stderr, "Error on file \"%s\" opening.\n", path);
		return;
	}

	while (getline(&line, &lineLen, file) != -1) {
		printf("%s|", line);
	}

	fclose(file);
	free(line);
}
```

Gives such an output, which is different than file content:
```
       ipsum
lorem  ipsum
```

PNieck, 2023-03-03:
Try without new line character after `lorem`

gerard5, 2023-03-03:
@PNieck I added a quick-fix PR for that, please check if this solves the issue before merging.

PNieck, 2023-03-03:
@gerard5 I have checked this PR and it seams to work

closed 2023-03-03 by gerard5