phoenix-rtos/phoenix-rtos-project #631: Lacks in setenv and unsetenv functions.

adamdebek, 2023-03-03:
1. According to Open Group unsetenv should return -1 and set errno to EINVAL when empty string is passed as argument.
2. Passing NULL as second argument to setenv causes exception.

gerard5, 2023-03-16:
@adamdebek I see in the edit history that the problem was also with `setenv()`
> ... Passing NULL as second argument to setenv causes exception.

but issue description has been redacted, you mention in the issue title that also `setenv()` lacks functionality. If it's about the second argument -- this is `well known` undefined behavior -- of course I can fix it like in glibc (see [a thread here](https://patchwork.ozlabs.org/project/glibc/patch/CALoOobNSbWUkd_i-L6U0ovbqPYnJY-h=ftX1K61yb19pmJj6aw@mail.gmail.com/#986239), similar problem), but than we should extend the docs, because "The Open Group (tm)" describes second argument very general:
> ..  The environment variable shall be set to the value to which `envval` points. ..
and
> .. The strings described by `envname` and `envval` are copied by this function. ..

So I believe that's why the issue description was redacted.

gerard5, 2023-03-16:
@adamdebek I've added fix for `unsetenv()` which will close the issue, please verify if it solves the issue, then I'll merge it if `setenv()` should be fixed in this pull-request as well please indicate it.

adamdebek, 2023-03-16:
Yes, you are right i will change the title of this issue, the only problem is with unsetenv() and setenv() should remain as it is.

closed 2023-03-17 by gerard5