phoenix-rtos/phoenix-rtos-project #633: race condition when using condWait to wait for interrupt handler

xvuko, 2023-03-06:
This is very common pattern used in our code to synchronize with interrupts:
```c
int irq_handler() {
    data_ready = 1;
    return 1; /* signal cond after exiting handler */
}

int thread() {
  mutexLock(cond_lock);
  trigger_irq_operation();
  while (irq_data_ready == 0) {
    ret = condWait(cond, cond_lock, 0);

    additional_checks();
  }
  mutexUnlock(irqlock);
}
```

If irq_handler is invoked after `irq_data_ready == 0` check and before condWait enters spinlock condWait will never return. cond Signal equivaled invoked from interrupt dispatch is ignored as condition queue is empty.  Using timeout could be a workaround, but it would be better to handle this on system/library level.

agkaminski, 2023-03-06:
Already solved: conditionals are "sticky", i.e. if empty queue is signalled, `cond` will remember that and exit `condWait` immediately on next invocation.

xvuko, 2023-03-06:
This is not reflected in phoenix-rtos-doc:
> The `condBroadcast()` and `condSignal()` functions shall have no effect if there are no threads
currently blocked on cond.

And not implemented in userintr:
```c
static int userintr_dispatch(unsigned int n, cpu_context_t *ctx, void *arg) 
        // ...
	if (ret >= 0 && ui->cond != NULL) {
		reschedule = 1;
		proc_threadWakeup(&ui->cond->queue);
	}

```

agkaminski, 2023-03-06:
It is implemented in `proc_threadWakeup`, so `userintr_dispatch` needs no additional mechanism to work.

IMHO this is an issue about documentation, as there's no race condition in kernel.

xvuko, 2023-03-06:
Yes indeed this is working correctly. I will follow with doc fix.

closed 2023-03-27 by xvuko