phoenix-rtos/phoenix-rtos-project #637: Scanf scientific notation floats problem

maska989, 2023-03-08:
### **During scanf tests there was a problem reading floats in scientific notation**

The problem occurs in an all-conversion specifier wich can handle that (%aAeEgGfF).

 - On reproducing this issue was used commit with sha number "bd57fdd3746e40dd60989277002ff949fcf0a348"
 - To reproduce issue copy following code:
``` c
	float test;

	printf("\n%f\n", 3.40282347e+38F);
	printf("\n%F\n", 3.40282347e+38F);
	printf("\n%a\n", 3.40282347e+38F);
	printf("\n%A\n", 3.40282347e+38F);
	printf("\n%e\n", 3.40282347e+38F);
	printf("\n%E\n", 3.40282347e+38F);
	printf("\n%g\n", 3.40282347e+38F);
	printf("\n%G\n", 3.40282347e+38F);

	sscanf("3.40282347e+38F", "%f", &test);
	printf("\n%f\n", test);
	sscanf("3.40282347e+38F", "%F", &test);
	printf("\n%F\n", test);
	sscanf("3.40282347e+38F", "%a", &test);
	printf("\n%a\n", test);
	sscanf("3.40282347e+38F", "%A", &test);
	printf("\n%A\n", test);
	sscanf("3.40282347e+38F", "%e", &test);
	printf("\n%e\n", test);
	sscanf("3.40282347e+38F", "%E", &test);
	printf("\n%E\n", test);
	sscanf("3.40282347e+38F", "%g", &test);
	printf("\n%g\n", test);
	sscanf("3.40282347e+38F", "%G", &test);
	printf("\n%G\n", test);
}
```
 - Outcome on Host-generic-pc and ia32-generic-qemu:
[image]

gerard5, 2023-03-08:
The `scanf(..)` implementation uses three functions `strtof()`, `strtod(..)`, `strtold(..)` to read and convert float/double/long double values. So the root problem is not in `scanf()` itself, but in the limited implementation of `strtold()` which is common base to the other two functions.

`strtoll()`, `strtoul()`, .. `strtold()`, `strtod()`, `strtof()` should be tested first as other functions like e.g. `scanf()`, `atoi()`, etc..  benefit from the implementation of the above-mentioned base functions so as not to "reinvent the wheel".

https://github.com/phoenix-rtos/libphoenix/blob/e913aa45b783bbc630dafac0ee36871a6a2324d9/stdio/scanf.c#L548
https://github.com/phoenix-rtos/libphoenix/blob/e913aa45b783bbc630dafac0ee36871a6a2324d9/stdio/scanf.c#L568-L580

damianloew, 2023-03-09:
Yeah thanks for noticing that, there already is such test (https://github.com/phoenix-rtos/phoenix-rtos-tests/pull/135). We will add similar cases to it and update issue with the code to reproduce using strto* functions. Applies also for https://github.com/phoenix-rtos/phoenix-rtos-project/issues/634

maska989, 2023-03-13:
@gerard5 I'm sorry for the lousy issue caused by a poorly selected available branch on libphoenix. Once again sorry for wasting time.